ncl-ROVers/surface #12: Create a skeleton front-end structure

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-21:
@hikarthik Let me know once you have an idea of the code's goal structure :)

hikarthik, 2020-12-23:
@TheCodeSummoner 

Finished setting-up developer environment on my system and created a new react app. Pushed the `react-materialui-app` repository outside the organisation and invited you as well.

A short summary of the things we discussed on the call would be helpful.

TheCodeSummoner, 2020-12-23:
@hikarthik Thank you - where is the repository?

hikarthik, 2020-12-23:
> @hikarthik Thank you - where is the repository?

`hikarthik/react-materialui-app`

TheCodeSummoner, 2020-12-23:
Okay, thank you.

I will keep an eye on it. Let me know once we have some basic src structure, and I can comment on it :) I will also get some of my industrial friends to suggest changes since I am not a front-end expert.

I'm trying to dig out UI design at this very moment btw.