ncl-ROVers/surface #13: Gather MATE related documents in one place

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-21:
@EldianTitan What do you think would be useful to share with the team?

My thinking is:

1. MATE Explorer PDF
2. Front-end design from previous years
3. Screenshots of current ROV model - might need to ask Rohan for those
4. Youtube links to some videos?

EldianTitan, 2020-12-22:
The front-end design from last year wasn't that complicated. I guess we can show them what it was like to give them an idea of what we want.

Screenshots of the ROV model sounds pretty good. However, we will also have the sim and if we get the current model in there I am not sure how useful they will be. That being said, we could give them the screenshots anyway since the sim will take some time to do. Same thing for the YT videos.

Not sure how useful the MATE explorer PDF will be. Only the people working on control systems might want it to read the task specs and it might just confuse everyone else (maybe?).

TheCodeSummoner, 2020-12-23:
I found this diagram on my PC: https://newcastle.sharepoint.com/sites/EngineeringProjects2021/Shared%20Documents/Systems/ROV%20components%20.html

I have also extracted the UI screenshots, such as:
[image]

Waiting for Rohan to upload some CAD screenshots.

Can you figure out where did our showcase video go @EldianTitan? I know I was flipping the ROV sideways in it - might be worth asking the society's committee.

Anything else left?

TheCodeSummoner, 2020-12-24:
Uploaded CAD screenshots to teams

TheCodeSummoner, 2020-12-24:
I couldn't find the video, but we can use our DEMO video for 2019 qualifiers - https://youtu.be/W3vK2PuRtQI

closed 2021-01-02 by TheCodeSummoner