ncl-ROVers/surface #14: Create a simulation repository

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-20:
@EldianTitan Are you available tomorrow (Monday) at 11.00 for a 30-min Teams call?

EldianTitan, 2020-12-21:
I think this is moot now since I missed it. Is tomorrow 11:00 ok?

TheCodeSummoner, 2020-12-21:
> I think this is moot now since I missed it. Is tomorrow 11:00 ok?

@EldianTitan Tomorrow is fine :)

TheCodeSummoner, 2020-12-23:
@EldianTitan I have reverted your commit. Like I've mentioned, please use a branch to commit your code rather than push to master, and submit a pull request when ready. I am expecting two branches to appear, one with source code, the second one with linting, testing, and CI infrastructure. I will not merge the former before the latter is merged.

EldianTitan, 2020-12-23:
Oh, ok. I thought you meant push the code to master and make another branch for testing, etc.