ncl-ROVers/surface #23: Get msgpack working on Arduino-s

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-20:
@ymber Do you think you can get some sort of simulator? Or maybe @antoinepetty or @EldianTitan have one of the society's Arduino-s they could share with you (Newcastle area only).

antoinepetty, 2020-12-21:
> @ymber Do you think you can get some sort of simulator? Or maybe @antoinepetty or @EldianTitan have one of the society's Arduino-s they could share with you (Newcastle area only). Similarly with Raspberry Pi for #29

I believe all of the society electronics are in the lab

EldianTitan, 2020-12-21:
I am not in Newcastle

ymber, 2020-12-21:
I'll look at simulation options. I don't know how much network code I'll be able to write before I know how the other end of the connection is going to work or what network hardware we're going to be working with.

antoinepetty, 2020-12-22:
> I'll look at simulation options. I don't know how much network code I'll be able to write before I know how the other end of the connection is going to work or what network hardware we're going to be working with.

It'll be serial UART communication over USB which is standard to all Arduinos as far as I know. (unless you decide to go with an alternative solution)
Sending data between the Arduino and pi has been solved already from previous years, it's just one byte at a time terminated with a newline character. (again, you have the choice to change this if you see fit)

ymber, 2020-12-26:
I understood from the meeting the other week that only the miniROV will be a wired connection. Do we not need wireless hardware for the main vehicle?

TheCodeSummoner, 2020-12-26:
Micro ROV is tethered to the Mother ROV. Mother ROV is tethered to the surface. You can have a look at the video I shared in Teams. Wireless communication doesn't work underwater.