ncl-ROVers/surface #24: Buy some hardware

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-25:
I propose to buy the following:

- Arduino Nano x3
- Raspberry Pi x1
- POE switch x1

We should have Pi-s in the lab as well but doesn't hurt to have more.

We have 3 Arduino-s, 2 are held by Joe Smith, and will be returned in the future.

@antoinepetty Can you tell me which exact Arduino Nano board should we get, as well as whether we need any other hardware?

antoinepetty, 2020-12-27:
> @antoinepetty Can you tell me which exact Arduino Nano board should we get, as well as whether we need any other hardware?

It'd be this model https://store.arduino.cc/arduino-nano-33-ble-with-headers (has an integrated IMU and Bluetooth which could potentially be used for debugging?)

We didn't finish the code conversation for this model. The new ID system still needs to be verified and it *should* work after that.

antoinepetty, 2021-06-09:
Ah this is where the information was!

We have purchased 2 of this model: Arduino Nano 33 BLE Sense
(one per enclosure)