ncl-ROVers/surface #26: Complete the photo comparison task

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-21:
Hi @NekTaf - would you mind posting your current progress here (some screenshots and a short description are fine)?

NekTaf, 2020-12-26:
[image]
[image]

NekTaf, 2020-12-26:
To-Do/ Improvements: 

1. Detecting when there is growth/death of coral and when there is bleaching or recovery, and applying appropriate contour color.
2. If the 2 images being compared are not perfectly aligned, ex. are taken from a different angle, then the code cannot detect the differences. 
3. Possible solution: draw contours and a "skeleton" of each coral, and focus on comparing the shape of the skeleton, rather than the exact dimensions of each coral, thus leaving room for errors, as the 2 corals cant always be perfectly aligned.
4. Remove Gaussian noise from image, isolation of key element/object (coral).
5. Code can detect colors, and draw a contour around them. However it cannot yet detect color differences between the 2 images.

TheCodeSummoner, 2020-12-27:
Are you following a certain algorithm or tutorial to complete this task?

NekTaf, 2020-12-27:
I'm looking through examples and tutorials on image processing and comparison. I need to do a bit more research though as a lot of the examples I found have the above problems, ex the images have to be perfectly aligned.