ncl-ROVers/surface #31: Schedule some meetings

TheCodeSummoner, 2020-12-17:


TheCodeSummoner, 2020-12-17:
Proposed meetings time - Friday every 2 weeks at 5 pm

mradzikowski, 2020-12-17:
I agree

Fisherchloe, 2020-12-17:
> Proposed meetings time - Friday every 2 weeks at 5 pm - 01/01/2021 first meeting

NekTaf, 2020-12-17:
> Proposed meetings time - Friday every 2 weeks at 5 pm - 01/01/2021 first meeting

Sounds good

ymber, 2020-12-17:
Fridays at 6pm would work better for me.

lestherll, 2020-12-17:
Fridays can work for me too preferably on the early evening starting at 5pm. I can't do 13th to 15th of Jan cause I have exams and deadlines on those days. I can't do 1st of Jan as well.

TheCodeSummoner, 2020-12-18:
Wednesday 06/01/2021 6 pm works for people? @lestherll @ymber @NekTaf @Fisherchloe @mradzikowski @EldianTitan

ymber, 2020-12-18:
That would work for me.

lestherll, 2020-12-18:
Works for me too.

mradzikowski, 2020-12-18:
> Wednesday 06/01/2021 6 pm works for people? @lestherll @ymber @NekTaf @Fisherchloe @mradzikowski @EldianTitan

It should work for me!

Fisherchloe, 2020-12-20:
Works for me

EldianTitan, 2020-12-21:
> 
> 
> Wednesday 06/01/2021 6 pm works for people? @lestherll @ymber @NekTaf @Fisherchloe @mradzikowski @EldianTitan

Works fine for me

TheCodeSummoner, 2020-12-21:
First meeting Wednesday **_06/01/2021 6pm_** confirmed then!

closed 2021-01-02 by TheCodeSummoner